gligli/p600fw #56: MIDI clock received during upgrade causes upgrade to fail

polluxsynth, 2015-01-30:
If a MIDI clock is received during an upgrade, this causes the upgrade to fail. While this is not something one would to knowingly, it is easy to have, say Cubase running, with the MIDI clock output set to Continuous, and try to perform an upgrade simultaneously.

I have not checked the actual MIDI data stream, but I would think that the MIDI clock, being a MIDI Realtime message, can be inserted in the middle of a sysex package, which the boot loader then cannot handle. The upgrade begins as normal, with U shown in the P600 display, but as soon as the upgrade sysex data is sent, the display changes to E. So the boot loader apparently properly waits for the sysex status byte, and it is later, once the sysex transfer has started, that the problem occurs.

Fixing it I suppose would not help existing units which are upgraded via MIDI, but should be noted in the manual anyway.

gligli, 2015-02-08:
Any MIDI message sent while sending a SYSEX firmware update frame would make it fail, but given that the MIDI parser while in update mode is really primitive, I think it's better to leave it as-is and just add a warning in the manual.

polluxsynth, 2015-02-08:
I agree, probably not worth fixing, but a warning in the manual would definitely help as it would set users along the right path if they cannot upgrade successfully.