gligli/p600fw #59: Sysex current program dump request?

hotswank, 2015-03-01:
I know holding FROM TAPE and hit PROGRAM 6 dumps ALL presets (which is very nice because this wasn't possible before), but I want to dump the current program bank only (by holding RECORD and hit PROGRAM 2, or sending F0H, 01H, 00H, 0ppp ppp, F7 in Logic Audio in the old Z80). This used to send the 16 bytes of the stored Non-Volatile program that is displayed in the PROGRAM display. Could you please put this back and let us know what the new dump request code will be, thanks!

PS will this dump all the extra new parameters as well? Could you please publish the new p600fw 2.1 program bit map? (like the old Z80: http://users.skynet.be/aemit/seq/p600.html). I'm re-building a my old P600 editor in Max and need to now the new program bit map.

gligli, 2015-03-07:
https://github.com/gligli/p600fw/blob/master/documentation/sysex_format.ods now has the info you need, it's still a draft, some things may be wrong. next alpha will support a SYSEX patch dump request too.

hotswank, 2015-05-07:
Disclaimer says "some things may be wrong", I asume you mean the "patch format identifier" [A5 16 61 00] and the "patch format version" [02]? The sysex dump in the latest may 2 alpha shows it's resp. [25 16 61 02] and [0]. Is this correct? It should have said "0x25 0x16 0x61 0x02" and "0x00"?

Furthermore I would exactly like to know what bit is doing what in MSB and LSB and what the max. values are, just like the original Z80 program bits map diagram from my first link. I would like to rebuild my editor in Max MSP and decode the MSB and LSB bits with objects like funnel, spray, peek and poke and convert them to decimals so the can reflect the current parameter values to the dials and switches upon receiving the sysex data (thank you for the single sysex program dump code btw!). The editing will be done by your dedicated MIDI controllers. I really hope you can provide me all the info I need and how the MSB LSB bits are organised. Thanks again!

hotswank, 2015-05-07:
[image]

gligli, 2015-05-08:
Dont forget the "Encoded data format" layer, it's purpose is to be able to pass 8bit values through 7bit MIDI SYSEX, you can't directly read/write patch data from the sysex dump.

Data is MSB first. All continuous parameters are 16bit and use the full range (0-65535) to represent paremeter value. Some parameters are dual sided and so the middle value is the center (32768).

hotswank, 2015-05-12:
I'm trying to figure out your encoder and it's really, really doing my head in :) First off, how do you count the 32 raw data bits and why do you repeat after only 4 Bytes? (encoded to 5 if you include the 32-23-15-7 padded MSB Byte) while normal MIDI data is send per 8 Bytes? 

From right to left?
Byte 0 - 31 30 29 28 27 26 25 24 ; Byte 1 - 23 22 21 20 19 18 17 16 ; Byte 2 - 15 14 13 12 11 10 9 8 ; Byte 3 - 7 6 5 4 3 2 1 0

From left to right? 
Byte 0 - 0 1 2 3 4 5 6 7 ; Byte 1 - 8 9 10 11 12 13 14 15 ; Byte 2 - 16 17 18 19 20 21 22 23 ; Byte 3 - 24 25 26 27 28 29 30 31

And how do you put back/count the encoded 7-bit bits (and order) when they are send? 
Because if you count the raw data bits from right to left and encode them, the first 7 bits of the 32 bit chunk of input data (bits 0-6), represent the 3rd Byte of the Patch format identifier 0x61, followed by 0x16 (bits 8-14), followed by  0x00 (bits 6-22), followed by the Patch number, followed by the MSB Byte which doesn't make sense if the first 5 Bytes of the encoded 7-bit sysex dump looks like this: 0 25 61 16 2 ... it looks as if the send encoded Byte order is left to right???

PS I'm trying to 'visualize' the encoder in an Excel sheet and will upload a screendump as soon as possible pending your reply.

hotswank, 2015-05-12:
The only way I can match the first 4 Bytes of the 8-bit raw data with the encoded 7-bit MIDI data is if the count order is like below because it's the only way 00h A5h 16h 61h 00h translates to 00h 25h 16h 61h 02h send by 7-bit MIDI MSB first... :

Byte 0 - 7 6 5 4 3 2 1 0 ; Byte 1 - 15 14 13 12 11 10 9 8 ; Byte 2 - 15 14 13 12 11 10 9 8 ; Byte 3 - 31 30 29 28 27 26 25 24

hotswank, 2015-05-12:
For the life of me, I can't figure it out, something doesn't add up.... and I really tried :)

[image]

[image]

[image]

gligli, 2015-05-12:
Ok, version number has been bumped between writing the spec and last alpha, hence why you get 7 instead of 2 as version. Version 7 adds microtunings per patch at the end of data, everything else is the same.

Seems like "osc a freq" is $fff0 in your example, it makes sense if the pot is fully clockwise (bytes are little endian ordered, ie the low byte of a 16bit word is stored first. I should probably write this in the spec)
That makes "osc a volume" $ff00 (fully clockwise, only 8bit pot scanning so the low byte always stays 0) and "osc a pw" $ffc0 (fully clockwise, 10bit scanning)

hotswank, 2015-05-12:
Is there anything else you haven't told us? LOL but thanks! Not sure if I understand now... so the only Bytes that are not MSB first are OSC A/B freq, OSC A/B vol and OSC A/B pw? Are you going to update the sysex_format.ods file in the master/documentation any time soon? Ooh and before I forget, are the switch bits i.e. LFO Mode Destination etc. specs notated in Hex or Dec? Would you like to publish your encoder code in Java/C++ or whatever? Thanks again!

gligli, 2015-05-12:
Data is all most significant bit first, but 16bit values are all encoded in little endian (ie least significant byte first, still those bytes are MSB), that's how it's handled by the avr architecture.
By the way, the fact that the lower bits are zero is not a requirement, just a concequence of how the pots are scanned.

I don't know but I need to clarify the 16bit values encoding and new patch format version yeah, as I said It's a draft, I made it because people requested it. The format itself is just a dump of the internal storage format, packed in a MIDI friendly way, not really something made to be easy to parse.

It's decimal, eg. "LFO mode dest" is a bitfield of those values.

Well, the code is in p600fw, mostly in storage.c and midi.c, but I think I'd better focus on improving the spec instead of implementing it in another language.

hotswank, 2015-05-13:
I have tried everything but something isn't right, sorry for my ignorance... Question remains how you count the bits in the input data stream and is the order of the last MSB Byte in the encoder correct (31 23 15 7)? Back to counting: do you treat each 32 bit chunk of raw input data as a 32 bits Byte or do you splice them up in 4 Bytes of 8 bits each?
 
[ 31 30 29 28 27 26 25 24 23 22 21 20 19 18 17 16 15 14 13 12 11 10 9 8 7 6 5 4 3 2 1 0 ] 
or 
[ 7 6 5 4 3 2 1 0 ] - [ 15 14 13 12 11 10 9 8 ] - [ 23 22 21 20 19 18 17 16 ] - [ 31 30 29 28 27 26 25 24 ]

Please take a carefull look again at my screen dumps to clarify the counting issue. As you can imagine, a incorrect counting order will skew the MSB Byte! I've have tried to project the sysex dumps from parameters OSCA freq 0 - 5, OSCA vol 0 - 99 and OSCA pw 0 - 99 and whatever I try, the 3rd MSB Byte (32 23 15 7) does not add up correctly! 

freq 4 vol 99 pw 99: 
60 10 4 0 7F 40 7F 0E
freq 5 vol 99 pw 99:
10 14 0 0 7F 40 7F 0E
freq 6 vol 99 pw 99:
70 19 4 0 7F 40 7F 0E
freq 99 vol 99 pw 99:
70 7F 0C 0 7F 40 7F 0E

hotswank, 2015-05-14:
I'm very sorry to be such a pain but I'm still trying to figure out how to count the 32 bits from raw input data into the encoder:

A) As one big chunk of 32 bits from R to L
Byte-0-1-2-3 [bit 31 < 0]

B) As one big chunk of 32 bits from L to R
Byte-0-1-2-3 [bit 0 > 31]

C) As 4 seperate Bytes, 0 starting at Byte-3 from R to L
Byte-0 [bit 31 <  24], Byte-1 [23 < 16], Byte-2 [15 <  8], Byte-3 [bit 7 <  0]

D) As 4 seperate Bytes, 0 starting at Byte-0 from L to R
Byte-0 [bit 0 > 7], Byte-1 [bit 8 > 15], Byte-2 [bit 16 > 23], Byte-3 [bit 24 > 31]

E) As 4 sperate Bytes, 0 starting at Byte-0 from R to L, repeat for every Byte from L to R
Byte-0 [bit 7 <  0], Byte-1 [15 <  8], Byte-2 [23 < 16], Byte-3 [bit 31 <  24] 

I'm pretty sure B) and D) are not valid because bits are always counted R to L [7 < 0].
Yet, if we count the raw data bits from R to L, as per A) and encode the first 32 bits chunk, starting with (bits 0-6), it represents Byte-3 of the Patch format identifier [0x61], followed by Byte-2 (bits 8-14) [0x16], followed by Byte-1 (bits 16-22) [0xA5], followed by Byte-0 (bits 24-30) Patch number [0x00], followed by the MSB Byte (bits 31 23 15 7), the sysex dump would look like this: 61 16 25 0 2

But if we look at the actual encoded 7-bit sysex dump, it says 0 25 61 16 2 ... so it does look like the 32 bits counting is L to R, as per B) unless the encoder also reverses the Byte order just like it does with the 16-bit WORD bits (little-endian).
If it doesn't switch Bytes and we look at the MSB conversion Byte of 0xA5 = 0x25, the only possible count order is C)...

In conlclusion: I really, really need to know the correct bit count order...

gligli, 2015-05-14:
If the raw input is 4 bytes in normal left to right order, eg:
A; B; C; D
Then the result should be, in normal left to right order:
0; A6-0; 0; B6-0; 0; C6-0; 0; D6-0; 0000; D7; C7; B7; A7

But you want to do the opposite, ie start from encoded data and decode it. So, this encoded chunk of 5 bytes:
V; W; X; Y; Z
Becomes, by reversing the process:
Z0; V6-0; Z1; W6-0; Z2; X6-0; Z3; Y6-0

Actually the spec was somewhat wrong, MSB and LSB ordering were mixed in the encoded data format, sorry for that.
I just pushed some fixes to the spec, including this.

hotswank, 2015-05-14:
Thank you very much! Another 3 days of my life gone heheh... but hey, you spend several months coding the new firmware so I will not complain! BTW Just d/l the new sysex_format and did notice the My Patch format version is still at 0x02 (should be 0x07) and the Modulation delay, Vibrato frequency, Vibrato amount, and Unison detune still say "unsigned" only, I guess they should be "unsigned, little endian order" just like the others?

gligli, 2015-05-14:
Patch format version 7 is from the latest alpha only, and only if the patch was written in that version, otherwise it's still version 2. For now it's better to use 2 if you write SYSEX patches yourself and just ignore the new stuff from v7.

I fixed those too, they are the same as the others now.