gligli/p600fw #60: Polyphonic sequencer

polluxsynth, 2015-03-09:
Instead of having monophonic sequences, they could be polyphonic, similar to the step sequencer in the Poly-800.

polluxsynth, 2015-03-09:
I've implemented a polyphonic sequencer now, it's still really only alpha, as I have not yet had a chance to test it on a real P600, only on my simulator in Linux, which operates via MIDI.

Anyway if you're feeling adventurous you can check it out. https://github.com/polluxsynth/p600fw/tree/polyseq

I was going to be more ambitious and have the possibility of having notes that did not necessarily all start and stop at the same time, as I believe the Poly-800 can do that. However, I ran into a problem with the transpose function, as it would mean having a note buffer covering the whole keyboard, where the value of the transposition for each note played would have to be entered, in order for the runtime transpose function to work properly. Since that would have required 61 bytes of RAM for each sequence, I opted to scrub that idea after some work on it and instead go with your suggestion of extending the existing format so that chords are stored as a normal events with an additional flag indicating that it's part of a chord.

I've changed the UI somewhat, so that there is now one UNDO/BACK (1) button and one STEP  (2) button. STEP works as TIE or REST did before, depending on whether it is pressed when a  key is pressed (TIE) or when no keys are pressed (REST). To me it makes note entry easier, as one can play rhythmically with one hand on the keyboard and the other on the 2 button.

The UNDO/BACK button (1) backs up one full set of events, i.e. a whole chord, or a tie or rest event.

Since my original idea would have required two bits per event, I remapped the keyboard codes so that in the sequencer they are stored as 0-60, leaving 61, 62, 63 for special events such as rests and ties, and bits 6 and 7 for auxilliary functions. As it is, bit 7 is the CONTINUE bit indicating that a note event is part of a chord rather than a new step, and bit 6 is unused. Perhaps useful for some future expansion.

I've also changed all the track length arrays in the seq struct to one large track struct, which in turn is put in an array in the seq struct. It makes the code easier to read and also seems to take up less space in the binary.

I've done some testing, but like I said, only in my simulator, so I'm hesitant to enter a pull request at this time.

gligli, 2015-03-11:
Cool! I haven't tested it yet, but I think we should include those changes for 2.1.
Will those changes still allow to play existing sequences? Not sure about old vs new internal tie & rests encoding.

gligli, 2015-03-11:
Well, there's a problem, the binary from your version is bigger than 65536 bytes, for now it can't be flashed using SYSEX.
The flash storage zone starts at offset 65536 and presets are stored at the beginning of it, there's actually about 24KB of unused flash in the middle of it, so I tried to move some code there, but no success for now...
Hopefully I can make it work, else adding more stuff to p600fw will prove very difficult, not that there's a lot left to add tho.

polluxsynth, 2015-03-11:
Ok, starting with your first comment, I haven't designed it so that it can play old sequences, if nothing else the note coding is different as I subtract 24 so that the notes are encoded with 6 bits, leaving to bits for flags. As it turned out, only one flag bit (SEQ_CONT) is actually used; originally I used both bits, one for note on/off, and one indicating that it's the start of a new step. Since the sequencer has not been in any official release yet I figured maintaining backward compatibility wouldn't be necessary. If you think it's important backward compatibility could probably be added with not too much trouble.

As for the second problem, I just got word yesterday that my P600 tester (i.e. my friend who owns a P600) had tried to flash the new version and the synth ended up acting very erratic. I was going to have a look at it this weekend, but good job you figured out what it was.

Is there anything I can do? Sounds as if the problem is related to the memory organization which I'm not very familiar with, certainly not on your level at this point. I could look at trying to optimize the code, but since it ends up being 67072 that's 1.5k too much and it would seem unlikely that it could be optimized so much that the code got cut down by that amount.

polluxsynth, 2015-03-12:
I was browsing through the code and I'm wondering if some space couldn't be saved by trying to replace  the float/double operations with fixed point operations using integers.

I did an experiment. First I temporarily removed all explicit calls to exp(), and measured the size of the text segment (exp() is still included as it is used by pow()). Then I replaced  pow(2.0, val) by (1UL<<val) in tuner.c (and foo*pow(2.0, val) with foo/(1UL << val)), making the whole pow and exp functions disappear from the binary. The difference in text size was 874 bytes, which is quite substantial. Admittedly, the exp() functions would have to be added back in somehow which will take up space. But I'm wondering how much could be saved by going on and replacing all float/double operations with fixed-point operations, realized using just integers.

cos() could be replaced by generating a cosine wave using basic DSP type operations, or a pre-calculated table if that is shorter. (The table is 256 uint16_t's long, which is 512 bytes, and it seems to store half a period of a sine wave (pi to 2*pi), but one could store only half of that and mirror it during init to get a complete half wave).

exponentialCourse is a tougher one. Would have to think about that.

gligli, 2015-03-14:
Ah well, I was afraid of if but I'm now sure there's a subtle bug in the updater code that prevents it from flashing anything above or equal to offset 65536.
The updater is is a chunk of flash that can't be live updated, so we're basically stuck with this problem forever if we want to keep the updating process as simple as it is today.

So yeah, I think the best option we have is to find a way to make the binary fit in 65536 bytes.

gligli, 2015-03-14:
Problem solved by commit https://github.com/gligli/p600fw/commit/0ce5d37d35395b698d24c3d5d14d820d71d0bba8 , binary down to 45KB :)

gligli, 2015-03-14:
I tested the new sequencer a bit, it's actually really cool to be able to have chords on a track, bass on another and then jam on it!
I found 2 relatively minor issues:
- You can add a single key many times to a chord, at first it works a bit like unison, but if you do it a lot of times, it ends up hogging all CPU ressources as you play the sequence.
- If you try to start both sequences at the same time by pressing both buttons at once, one of the sequences often gets delayed by one step, odd.

polluxsynth, 2015-03-14:
Great about the -O2, I never even thought to check the compiler options (or if I had I might not even have dared change it as I remember how the code failed to run properly with optimizations switched off completely).

Glad you like it. As for the two issues, the first one is something I forgot, I was planning to avoid adding a note to a chord if it was already to be entered, to avoid this situation, as the resulting unison would not be something one naturally would think would happen. Rather if one removes notes from a chord, the idea is that the user has pressed a note by mistake, so if a note is pressed again it would be because he realized he had in fact not made a mistake and did want the note to be inserted.

I'll have a look at that as soon as I can. I wonder why it hogs all CPU resources though, it should just send a bunch of note ons to the synth which shouldn't be different from them arriving via MIDI from, say, a master keyboard.

The second one I've not been aware of, thanks for mentioning it, I'll have a look and see if I can figure out what's happening.

polluxsynth, 2015-03-15:
I've now tried to fix the two issues. It's now not possible to add duplicate notes to a chord. I haven't limited the number of notes though (useful when playing an external sequence), so I don't know if the same CPU hogging will happen if lots of (different) notes are entered at once anyway.

The second one was rather intricate. I think the problem was there also in the monophonic sequencer. I've tried to explain it in the commit message and also in the code; basically, the root of the problem is that seq.conter was set to INT16_MAX whenever a new sequence was started, which means that if both sequences are started at the same time, as soon as seq_resetCounter() happens for the second one, it will cause the first step of the first sequence started to be cut short. Since both sequence buttons are pressed virtually simulatenously from the user's point of view (but sequentially from the software's point of view), the first note of the sequence which happens to be first is cut short and in practice not heard at all.

Anyway, please give it a try, I've re-pushed my polyseq branch referenced above (do note that I've modified the "alpha" message in the firmware to reflect the fact that it is not one of your alphas) with the new changes. Again, so far only tested in my simulated environment.

gligli, 2015-03-15:
Seeing the code, I can't explain it but it seems it can't get out of record mode now. Hitting play sometimes even corrupts RAM (garbled displayed text).

polluxsynth, 2015-03-15:
It's not something basic like the fact that I git rebased my polyseq branch bringing in your latest changes on master (-O2 in the Makefile) so that you need to do a complete make starting with make clean?

polluxsynth, 2015-03-18:
Did commit b7dc537e24bb4907859a8e46eac82228c3498588 and/or 6504831ce84edb74a653337f12db54f44e594ad2 solve the problems with the sequencer?

gligli, 2015-03-19:
Ah, I don't know, I'll have to check, that problem I had could be related to storage yeah.

polluxsynth, 2015-03-19:
I'll try to lay my hands on the P600 for some testing here, it might take a week or so though.

polluxsynth, 2015-04-04:
Finally got the P600 home for testing. I'm working with my polyseq branch:
https://github.com/polluxsynth/p600fw/tree/polyseq

I can't get out of record mode either. I have debugged it so far as the variable tp in seq.c:seq_setMode() on my branch, which is set to point to the track struct for the track in question. It seems that calling storage_saveSequencer() changes the value of the variable?! In the particular build I'm working with it is 0xEEF before the call and 0xE00 afterwards (determined by saving the value of the pointer in a static variable, and comparing it after the call, then sprintf:ing the values to a string and sevenSeg_scrollText:ing it if they differ). Removing the call to storage_saveSequence() causes the bug to go away (but of course the sequence is not saved then).

This seems very odd to me. The two possibilities I can think of now is that either there is a compiler bug which for instance causes the registers that tp are stored in not to be saved properly across the call, or there is some function called by storage_saveSequencer() which either corrupts the stack (if tp is on the stack) or corrupts a register it shouldn't touch (if tp is in one or more registers). Either way it's strange it doesn't happen anywhere else in the code. Of course very few functions actually save something to storage, so if it is something called by storage_saveSequencer() (e.g. something in teensy_bootloader_hack.h) it would only be visible to a few functions.

polluxsynth, 2015-04-04:
I don't have the P600 set up so I can easily reflash the bootloader so I can't test this at the moment, but looking at teensy_bootloader_hack.h, in blHack_page_fill there is a movw r0, %4, yet of the register pair r0:r1 only r0 is listed in the clobber list in the __asm__ statement.

gligli, 2015-04-05:
The code in teensy_bootloader_hack is mainly a copypaste of a gcc file iirc, the difference being that istead of calling "spm", here we have to call a piece of the bootloader.
I'm thinking that call to teensy bootloader does more than just spm, maybe it thrashes some registers too.
I think the best thing to do is to store all regs, eg to stack, before the teensy bootoader call and restore them after. Or maybe in storage_write() so that it can be effective when updating thru SYSEX.

I'm currently moving out so I can't work on p600fw, I should be back on it maybe next week end.

polluxsynth, 2015-04-06:
Yes, I was wondering if the SPM call does something else apart from SPM too, or if the address to it is off by a couple of bytes so it does something unrelated before the SPM? How has the SPM call address been determined? I would think it would be possible to read the code at that address to see what is really going on?

Anyway, I added an __asm__ statement to storage_write which effectively just tells the compiler that R16 is clobbered, causing it to push and pop that register in that function. This cures the problem. You're right, perhaps more registers should be pushed. Still, I could tell that R17 (the high byte of R16:R17) was unchanged through the call anyway, so I'll just leave it at R16 for now.

I've pushed the new version to https://github.com/polluxsynth/p600fw/tree/polyseq .

I've also attempted to fix the problem of the arpeggiator and sequencer not running in sync (they run at the same speed but the note ons are synchronized to when the sequence was started, or note pressed when it comes to the arpeggiator, rather than each other when both are running). It's at
https://github.com/polluxsynth/p600fw/tree/arpseqsync . There's still a bit of work on it needed, as the arp and/or seq get out of sync with the beat as they reset the clock as soon as the respective button is pressed, which I believe should not happen in MIDI sync mode

closed 2015-05-02 by gligli