gligli/p600fw #61: Sync to midi clock

mink99, 2015-03-15:
The values that can be set for the clock divider for the arpeggiator and sequencer range from 0-127.
As midi clock is set at 96 ticks per beat, I would suggest either 
... To give it a range from 3 to 96 so that the given entry would be the divisor from 96 as 
3 would be a 32th note, 6 would be a 16th, 8 a triolic 8th etc...
... To give it a fixed range of values : 1/1 1/2 1/4 1/8 1/8t etc

It would be an enhancement to have this divisor stored globally and not per patch, as patch switching in a sequence creates unpredictable rhythmic results.

Sometimes, when a sequence is stopped, the note-offs are not send, therefore we have hanging notes. A short "all notes off" when stopping a sequence would help.

The sequencer still seems to be some kind of jittery, maybe when the divisor can be set in a documented way, I would like to offer my help in testing this

Regards Kris

gligli, 2015-03-15:
There's already a fixed range of values, the possible divisors to midi clock are: 384,192,168,144,128,96,72,48,36,24,18,12,9,6,4,3,2,1

Storing it globally doesn't seem like a bad idea. I could also remove 2 divisors to make the 0-127 range nicely split in 16 divisors.

Hmm, which alpha are you using? There was problems with hanging notes in old alphas but I thought it was fixed in recent ones.

The sequencer is updated at 500 ticks per second, so worst theoretical jitter would be 2ms. Some actions could make it worse tho, eg loading another patch or changing settings.

mink99, 2015-03-15:
Hi, 
I am using the http://sfx.gligli.free.fr/zik/p600/dev/2015-02-22/p600firmware.syx. Version. The note hangs occur when the external clock is stopped, tested on a motu xl midi interface, which is reliable, so that we are not chasing ghosts... 

It sounded as if the notes were shuffled, the test sequence was a 
1 1 1 pause, independent of the resolution or tempo  it seemed as if the second and third step were shifted towards the beginning of the beat.  I will recheck.
There might have been a pebkac on my side, as I could not find a match between the 127 steps display and the divisors. 

I am currently planning to build my own step-sequencer on arduino base , also with the poly-800 in mind. In this concept the useful steps are much less : 1/32 1/16 1/16t 1/8 1/8t 1/4 1/2 1/1 .  Where I am sure that even 1/32 notes will have issues due to the still- not- so--fast envelope generators.

One last thought, there are devices that permanently send midi clock and assume the synced devices to start after a midi start 0xFA and stop after a 0xFC. Some devices ignore this (volca) other respect it (mpc). I am not sure which approach would be better...

Thanks 

Kris

gligli, 2015-03-15:
Ok, I just checked my git log, I commited a fix for that just after releasing 2015-02-22.

As I said, there is probably a bit of jitter, nothing that should be obviously out of sync tho.

I already made some changes to the way clock is stored, I chose to limit it to 16 divisors, removing the fastest and slowest ones and made it a global setting.
On the SW side, envs are as fast as they should be. On the HW side, aging S&H 4051 multiplexers can have a hard time following them, there's nothing I can do about it.

I think I just ignore clocks out of a start/stop right now.

mink99, 2015-03-15:
Please do not get my words on the envelope speed as critics on your work , this was never my intention. ... And sorry, I saw your last commit only after I had placed my last comment.

I will retest the next update / alpha and report back if that is ok for you

Regards Kris

gligli, 2015-03-16:
Hey, no problem, I just wanted to explain the reason why envs may still be slow.

Ok, next alpha will have lots of new stuff, probably including polyphony for the sequencer.

gligli, 2015-09-06:
I'll assume this is solved.

closed 2015-09-06 by gligli