gligli/p600fw #67: Pressing TUNE causes synth to emit a short blip

polluxsynth, 2015-04-18:
Pressing the TUNE button causes the synth to emit a short blip on the audio output. This seems to be due to the final volume VCA not closing fast enough (the time constant on the RC is in the region of 30 ms).

polluxsynth, 2015-04-19:
One thing I didn't understand when looking through tuner.c, is the way that DAC updates seem to happen very seldom. For instance, in tuneOffset(), sh_setCV() is used with the 'immediate' flag set to 0, so that the CV is actually not output until whileTuning() is called from measureAudioPeriod() (which admittedly is called very soon after). Is there a reason for not doing an sh_update() immediately when the CV has been set? Also, whileTuning() is among other places called from a loop in measureAudioPeriod(), which means that it continuously writes to the display during this time, which also seems unnecessary. This also means that it writes multiple times to the DAC by calling sh_update() in whileTuning(), is that to get a stable DAC value, or is it in fact to add a bit of delay, as there is a comment in measureAudioPerod() about that 25 loops with calls to whileTuning() is necessary to get the filter tuning working in a stable manner.

gligli, 2015-04-19:
Only one third of the display can be lit at the same time, hence the need to refresh it continuously.
The tuner is highly sensitive code, easy to break something on some P600s, I would recommend making the least possible amount of changes in it (you fix for the initial problem seem perfectly fine tho).

polluxsynth, 2015-04-19:
I don't really want to make any changes, mostly curious as to why it looked the way it does.

commit 1aece145 (2015-04-30): Merge pull request #73 from polluxsynth/release

Fix #67, #64, #43 and #69
closed 2015-04-30 by gligli