gligli/p600fw #70: Sysex messages don't seem to be receiving

snickell, 2015-04-22:
I think I have the "midi tuning standard" PR updates against master, but master seems to have had a regression on receiving sysex messages. I tested against master from a year ago and sysex messages do receive correctly.

Every time I try to transmit sysex to the p600fw, I'm seeing what looks like memory corruption somewhere in midi.c (and end up getting a "sysex message overflowed"). 

Any idea what commits might have affected this? I'll work on tracking it down, but its the sort of bug that's a little tricky w/o jtag!

gligli, 2015-04-22:
A memory corruption could occur because too much RAM is used for data and there's not enough left for stack, so do you think your changes use more RAM than master?
I know we're kind of short on RAM that's why it made me think about it.

snickell, 2015-04-22:
The issue occurs on unmodified gligli/master, so not something related to
my changes. There's some fairly convoluted pointer arithmatic going on in
midi.c, esp dealing with filling the sysex buffer, so I'm suspicious of
that area of things.

Unrelated thought: I don't know the specifics of the avr architecture and
how avr-gcc allocates memory, but I thought stack was usually a fixed size
allocation (so from a different memory pool than malloc which means 'using
more ram' shouldn't affect the stack unless you overflow the stack in a
particular frame).

snickell, 2015-04-22:
Sysex's receive correctly on d83dd7e (Oct 2013), but do not receive on current master b311b96 (Apr 2015). If I have time I'll try to isolate the commit that caused the regression.

gligli, 2015-04-22:
Another wild guess, it could be this commit, it made the sysex buffer smaller: https://github.com/gligli/p600fw/commit/7224ec8071e5dc2ac1f2e6e1ce614ce6118c001c

snickell, 2015-04-23:
> smaller: 7224ec8

I think that just changed the storage buffer, but I'll check if
before/after commit made a difference.

snickell, 2015-04-23:
Argh, my fault. I've had my midi in/out cables reversed, but the old 'echo' kept me from ever noticing :-P Sysex working fine, my fault entirely.

closed 2015-04-23 by snickell