gligli/p600fw #74: Tuning issue

mwooten1, 2015-05-17:
Hi. Just some background info: This is my first vintage synth. I purchased it online anout a month ago from someone who had already installed the 2.0 update. It has been working perfectly until today. I was playing it and it drifted a little out of tune. I pressed the tune button and then afterward it was much worse. Now only one note (usually middle c but it drifts down to b or bflat) is in tune and everything around it drastically out of tune. Like not just really flat or sharp, if you play a chromatic scale straight up you just get two different random pitches for each note. I've tried everything I could think of and read the manual all the way through no luck. Any advice?

mwooten1, 2015-05-17:
The only thing that I did a little bit before the problem starter was kill some of the oscillators. This was my first time doing this so maybe that did something I don't know.

gligli, 2015-09-06:
Sounds like a hardware problem if it started acting without any firmwmare change. A failure related to the DAC could do this...
One simple way to rule it out is to try to do a full sweep with a pot, if the display doesn't jump / misses values, then that's probably not it.

Sorry for the late answer in any case, feel free to reopen if you think it's a software problem.

closed 2015-09-06 by gligli