gligli/p600fw #77: Raw patch data format parameter swap?

hotswank, 2015-05-30:
Is it possible to 'relocate' the position of the 'Modulation wheel target' (incl. padding)? It would be better if it would come right after 'Modulation wheel range' andbefore 'Osc pitch mode'. This would not only be a more 'logical' position e.g. clustering similar parameters like Pitchbend, more importantly, this would 'push' the 2x8 'Modulation delay' bits in the 'next' cluster of 4 Bytes of the encoded data string. I'm not sure why you didn't do this, but perhaps you had a very good reason for it?

Below is a picture of the 'before' and 'after' situation to get an idea what I'm on about!

[image]

hotswank, 2015-05-31:
While your at it :) would it be possible to add an extra padding byte at the end (before the 6*8 'Unison track pattern' bits) so bit 7 of 'Arpeggiator speed' is isolated from decoding the MSB byte of 'Unison track pattern' (and thus can be ignored). This will also allow me to visualize the 'Unison track pattern' as notes more easily.

[image]

gligli, 2015-05-31:
Actually I can't make any change to this format, it is basically how a patch is stored in flash, and it must be kept backwards compatible.
The format was extended for 2.0, so new parameters had to be added after the old ones, hence why some parameters are not where they should logically be.

hotswank, 2015-05-31:
Now that's a real shame :)!