gligli/p600fw #79: P600 GliGli Control Panel Problem

polylux, 2015-09-26:
Dear Fabrice, the software worked fine for a few weeks, but from one day to another some strange control panel behavior occurred: 0 changed the place to 8, means by pressing 8 the display shows 0, the 8 dissapears completely, pressing sequence one now gives access to arpeg speed, completely no function by pressing 0, no access to all related 8 functions, all other numbers and functions are as should - I loaded stable 2.0 and all updates since then but the problem is still there, I don't think it's the panel itself, do you have any idea? Thanks in advance, Roland

gligli, 2016-06-23:
Hi, it doesn't sound like a software problem if it worked fine at first but the only way to be sure is to try with the old Z80. If it still acts strange with it, then it's a hardware problem.

polylux, 2016-06-23:
Hi Fabrice ! Better late than never ;) We talked about a few months ago on gearsluts, and yes it was the faulty membrane with corroded contacts - problem solved - All the best, Roland

gligli, 2016-06-23:
Haha yeah sorry :) Didn't allocate much time for this project in the last months so I totally forgot about it.