gligli/p600fw #80: Original Sequencer of P600 not working anymore with Upgrade?

albanpao, 2016-01-09:
Hi, just installed the new upgrade (hardware and software). I noted that the original realtime polyphonic sequencer is not anymore active... this is a normal consequence of the upgrade or something is going wrong?

jointtherapy, 2016-06-06:
My sequencer isn't working at all either. Does anyone have any updates of this issue? Thanks

thorik1961, 2016-06-23:
Got same problem 
Can anyone help

gligli, 2016-06-23:
The sequencer hadn't been remade yet for release 2.0.
However, release 2.1 will have a new step-based sequencer.
2.1rc3 is already available there: http://gliglisynth.blogspot.fr/2015/09/p600fw-210-third-release-candidate.html
The manual hasn't been updated yet for 2.1, hence why it's still in rc. In the mean time, previous blog posts can help to understand how the sequencer works.