gligli/p600fw #84: Update the manual to reflect changes made from 2.0 to 2.1

gligli, 2016-06-23:
My blog posts should help for this: http://gliglisynth.blogspot.fr/search/label/p600fw

numind06, 2016-11-18:
please contact me gligli, I can't leave any message via github. skype: numind06 or mail: dj2jm at hot...com

sorry for disturb.

regards.

gufzzz, 2020-08-19:
Hi gligli, I would be happy to make an update on the user manual, just made an editor un Cubase and would love to get info on the new parameters available. Contact me if you are interested.
[image]

hotswank, 2020-08-21:
Hi gufzzz, could you tell me/us what CC59 values you used for each LFO mode destination? I'm finding it difficult to pin point each CC value for each combination? When I go from 0-99 with my midi fader I can hear the LFO destination changes and they also repeat themselves after a while but nit exactly. I would like to know which value is what, i.e. Filter, VCO PW, VCO A+B etc. Thanks!