gligli/p600fw #86: Tuning failures with 3340 Rev G chips

eframp, 2016-07-17:
Now that Curtis/OnChip has re-released the 3340, I've been able to re-populate an otherwise dead Prophet 600 (all the 3340's had been harvested out of it). I've done the resistor swaps as specified in the service manual to bring the tuning trims back toward center.

When using the Z80 scaling routine, I can scale the oscillators perfectly
When using the Z80 Autotune routine, the unit tunes the oscillators perfectly.

When using the GliGli scaline routine, I can get within +/-3 on the display, so it seems to be working fine.
BUT, when using the GliGli AutoTune routine, 11 out of 12 oscillators are tuned incorrectly in the three octaves below Middle-C pitch (as in, if I transpose the oscillators up or down on the keyboard, the scaling is still completely incorrect below the pitch of middle C, not physical middle C on the keyboard). I can't find anything electronically different about the one correctly-scaling oscillator (it's osc. 2B, FWIW).

I've tried two different Teensys flashed with various versions of GliGli (1.0, 2.0, 2.0RC3) and am getting the same results, so I think it's fair to say it's not just GliGli or the Rev G chips, it's the combination of the two. The unit is functioning perfectly (both with Z80 and with GliGli) otherwise.

eframp, 2016-07-21:
Ace Sequential tech Greg Montalbano also had a P-600 with Gligli and a combination of original and Rev G 3340 chips on the bench, and suggested the following workaround. When scaling the instrument, completely ignore both the GliGli and Z80 tuning routines, and set the oscillator trims back to about 12:00 on the dial, as you're looking at them from the keyboard.

It took me a couple of tries, but sure enough, it works.

bad-pritt, 2019-04-28:
Hi, I'm a late Tensi Gligli user, and I'm looking for some help. Here's my problem: first everything went fine but after one day the tuning function sometimes did'nt work (no light, no action), then the two sequencers  and the second arpegiator stoped working. I tryed to update the firmware but the little button on Tensi jumped out ! And the orange light did'nt lite anymore... nowadays the two sequencers works again but I still can't use the two sequencers nor tune my Prophet. (yes, complicated crash story)
Does anyone knows how to solve these bugs please ? Thanks a lot !