gligli/p600fw #87: CC's not working correctly

MattDiamonds, 2016-10-16:
Hi everybody,

1st of all thanks a lot gligli for your work, it's a huge gift you did for the P600 lovers ;-)

I've got an issue with CC's as it seems to work or notwork depending on the position of original pots on the computer.

Check this video, you'll see that the Cc22 works for Cutoff, then i move the real cutoff pot, and the CC22 doesnt work anymore, then i move the P600's Reso pot and the Cc22(cutoff) works again...

https://youtu.be/A345M8rOyys

Somebody (Gligli?) could help? 
May be using Nrpn? Dont'k know how to setup but the Behringer bcr2000 seems to be able to manage nrpn's.

As a last info, i had the exact same issue with an other Cc Controller.

hotswank, 2016-10-16:
This is 'normal' behavior and of technical origin. Nothing to worry about but yes, a little annoying.

Regards,
Maurits

Johnnydiamond, 2016-10-16:
Is there an update for the sequencer? My gligli P600 doesn't work.

-

gligli, 2016-10-16:
It may bave been fixed in 2.1rc3, but I'm not sure, it's been a while since I worked on p600fw code.
2.1rc3 also contains a step sequencer, downloads are there: http://gliglisynth.blogspot.fr/search/label/p600fw

MattDiamonds, 2016-10-16:
@gligli thanks for your quick response, I did install the rc3 before asking ;-) And i'm sad the same issues occurs...

Do you think there's an other solution or i have to abandon the idea of controlling by midi?

My project is to make a 19' Rack of the P600, without keyboard & knobs, for the stage concerts i do. Just to play it in midi with my microkorg XL+ that has lot of custom CC possible to send.

... ;-)

gligli, 2016-10-16:
Now I remember the last tweaked panel pot does parameter changes continuously to avoid stepping that would be caused by change detection. So it's effectively a technical choice.
As long as you don't edit a parameter by both MIDI and panel, it should work fine.

MattDiamonds, 2016-10-19:
Thank you all guys for your replies, specially @gligli for all your efforts building this and remembering all the process to respond ;-)

Take care

MD