gligli/p600fw #88: Nothing at all happening?

bradholland, 2016-10-24:
Hi there, 
I'm a reasonably experienced tech. And have followed instructions to the letter.
The teensy programs just fine, and I have tried two different teensies.
The synth works just fine on the original Z80 but appears dead when the teensie is installed. 
The power supply has been recapped. 

When I turn the synth on with the teensie installed, its led lights up for about 1-2 seconds then goes off. 
Having cut the 5v trace on the teensie and removed the battery, I can program the teensie in situ seems to program fine, but still nothing...

Does this sound like any familiar issue?

thanks in advance for any responses.

bradholland, 2016-10-24:
I should point out, that when I mean dead, the control panel doesnt light up or do anything.

bradholland, 2016-10-29:
moral of the story is, don't by chinese clone teensies...

gligli, 2016-10-29:
Ah yeah true, teensy clones, eg. from EBay don't work with p600fw...

closed 2016-10-29 by gligli

bradholland, 2016-10-29:
it may be that they have an arduino bootloader?

bradholland, 2016-10-29:
in any case, works just great with an original.

gligli, 2016-10-30:
Probably, I use part of the teensy bootloader to save stuff to flash.