gligli/p600fw #90: VCA Release Issues on p600

blackshapeswhitenoise, 2016-12-19:
Hi Fabrice and forum,

This is most likely more of a hardware problem, but I'm just checking from the software point of view here.

My p600 is experiencing abrupt voltages drops to zero around the end of long envelopes on all six voices. This is most likely also happening on shorter envelope release settings but it is not as noticeable.

Attached are some pictures of oscilloscope readings showing the drop off.

I just want to make sure that it's definitely not an issue with software before I start replacing things like the op amps, demultiplexer, capacitors etc. For example, maybe the hardware and software are working fine, but I might need to make an adjustment on the DAC trim (I have the DAC700 with trim circuit). If it was the computer generated envelopes that were the issue, is it that they just reach a point where they turn off?

Just to clarify, I made these measurements at each voice's S/H circuit after the op amp.

If anyone on the forum has any advice, it would be much appreciated! Thank you!

[image]
[image]

blackshapeswhitenoise, 2016-12-19:
Somehow the photos came out upside down. Sorry!

blackshapeswhitenoise, 2016-12-22:
Just updating here and realizing that you actually acknowledged this issue on a 2.1 alpha post from 2015-2-08, I loaded stable release 2.0 and the issue is no longer there. I just want to let you know that issue is on all the final release candidates and alphas up until 2015-2-08 at which point I just assumed it was on all of them.

Did we ever get a workaround for this that I missed or is it still in the works?

Thank you very much to who have contributed to this fw thus far.

gligli, 2017-01-15:
Hello,
My P600 had the opposite problem, ie envelopes ended too soon and it seemed like there was a range of VCA voltages that just gave a fully closed VCA so I added a small voltage offset to fix this.
Now it seems like your P600 has a "perfect" VCA so that fix has the opposite effect of not fully closing the VCA...
I could compile a 2.1 with that fix removed and it VCA should behave like 2.0.

blackshapeswhitenoise, 2017-01-15:
Thanks for your reply Gligli!

To clarify it seems like my envelopes are closing too soon as well. The upside down pictures might be closing too soon as well, as if the envelopes were ending or being truncated. Sorry if the upside down photos were misleading.

Since I've last posted I was looking into the source code and I'm assuming the offset has something to do with the VCA DEADBAND constant right? I don't really know much about programming, I would definitely love to try a version without the offset as well as one with a different offset.

I also would like to note that when checking my Vdac output voltage the way the service manual recommends by measuring the output of the s/h for Mix A, I am getting 4.92V. Maybe the calibration of the Vdac trimmer is more important and critical with the new firmware?

Happy new year, Gligli! It's nice to hear from you!

gligli, 2017-01-15:
Ah ok, so you have the same problem I had, just worse. In that case, yeah, VCA_DEADBAND needs to be set higher.
If you can get a voltage closer to 5V by adjusting the Vdac trimmer, it should help the whole synth work better, not only the VCA.

Here's a 2.1rc3 version with the VCA_DEADBAND doubled: http://sfx.gligli.free.fr/zik/p600/2_10_rc/2_0992_VCADB/p600firmware.syx

blackshapeswhitenoise, 2017-01-17:
Thanks for the firmware!

It made the voices more uniform with each other, but perhaps made the
release even more abrupt. The release now sounds like a hard cut once it
reaches the end of its envelope.

I will put back the last Alpha back on and try to adjust the VDac trim to
see if it helps.

Thanks for your continuing help!

blackshapeswhitenoise, 2017-01-17:
After adjusting the trim on my DAC, I realize that the trimmer only let's
me get as far as 4.93V.

It actually didn't seem to make too much of a difference as far as the
release envelopes went. As such, instead of going closer to 5V I went back
to the service manual recommendation of 4.9V.

The behavior was seemingly the same at 4.9V, 4.92V, and 4.93V so I guess
the service manual is right in stating that it might not be very critical
maybe? Debating whether to keep it at 4.9 or 4.93 currently.

gligli, 2017-01-17:
Ok, so you do have the opposite problem I had, which makes sense if the VCA_DEADBAND fix was what made your P600 behave wrongly.

So, this version which has VCA_DEADBAND set to zero should fix the problem: http://sfx.gligli.free.fr/zik/p600/2_10_rc/2_0992_VCADB2/p600firmware.syx

blackshapeswhitenoise, 2017-01-18:
> p600firmware.syx

It seems like this has fixed the problem.

I thought at first that maybe each voice was having different release
times/curves, but I let my ears rest and sit awhile while the synth warmed
up and they sound good. I also checked with my oscilloscope and it's
looking good as well!

Here is a photo of fast exponential.

[image: Inline image 1]

So if you put up another final release candidate, or release stable 2.1,
might there be an option to disable, redefine, set the VCA DEADBAND? Just
in case there are more people with the same issue as me or if the feature
wasn't there, for people with a synth like your own?

As always thanks for all your help, I really appreciate it!

rundgrenrules, 2021-08-29:
Does anyone happen to have either of these DEADBAND adjusted firmware files available?  I have had the same issue across all revs of the firmware and would LOVE to fix it!