gligli/p600fw #92: Ploblems after failed upgrade

prohet13, 2017-01-22:
Hi folks, as the teensy board (https://www.pjrc.com/store/teensypp_pins.html) is out of stock I just ordered one on ebay where I had to solder the pins by myself onto the board. After I had done this I followed exactly the steps from gliglis document but with no result. The p600 didn´t work after I finished my work which is caused by my bad solder skills I guess (looks really terrible). However, I dicided to go on with the original cpu and come back later with an new, pre solderd teensy. So here is my problem:  when I put back the z80 into the socket and powered on the p600 there was nothing on the display. I tried once again and again.. sometimes appeared a permanent noise and some undefined symbols, sometimes it happend nothing -.- 
No button or poti shows any reaticon as well.

May I destroyed the solder joints from the socket when I first removed the z80 (it was really hard to get it out) ? Or is there just any procedur to re setup the p600 after removing the cpu ? 

I would be really happy for some help cause I have no idea what to do now with my lovely p600 :( :( 

Thanks and kind Regards

gligli, 2017-01-22:
Hello,
It's hard to tell what's wrong but teensys from EBay are often chinese clones and don't work with p600fw.
Going back to the Z80 should just work, not additional procedure.
I'd say double check your soldering on the teensy, if everything was correct, then yeah, the Z80 not working was probably a dry solder joint or a Z80 socket bad contact, something like that...

prohet13, 2017-01-22:
Hi, thanks for the fast reply ! 

I found the problem: because of the bad prepared teensy there were two contacts from the socket bent so they had no contact anymore to the z80 anymore. Now I fixed this and the synth is working.. glad that there are no problems with solder joints on the card.. 

The teensy I orderd is this one: http://www.ebay.de/itm/322139808714?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

Maybe it didn´t work because this teensy is bad but I am almost sure that I destroyed some contacts on the board as well while treating it with very bad equiptment and no skills *grrr* 

Is there any option to get an pre soldered teensy ? At least with pre soldered pins ? Or do you have any idea how long it takes until the stock from pjrc.com is refilled? 

Thanks for your help !! Kind Regards

gligli, 2017-01-23:
Ok, good!

Looks like a genuine one from the pics. I think so far all teensy clones I heard of had a blue led instead of orange.

No idead about PJRC stocks, but I think you can get teensys from EU shops, look at this thread on pjrc.com: https://forum.pjrc.com/threads/23601-Official-Distributors.

Winbyte, 2017-01-24:
MY first teensy was from china  and it did not even take the Flash from the teensy loader!

prohet13, 2017-01-25:
I finally got a new teensy from Watterott Electronics and with some new solder equipment I made the upgrade successful :) Thanks for your help and for your awesome work in general !! I am really excited about the new power the synth provides now :+1:

prohet13, 2017-01-25:
One more question I have :

 with the numerical values from the display I see that the bender isn´t in zero position when it latch into the middle of the track. I also had problems before when I tried to do the tuning right with some other synths.. now I know where it comes from.. So do you maybe heard of this issue ? I think its just a problem caused by the hardware ?! Can I just fix it by turning some screws ?

gligli, 2017-01-25:
Cool, good to head it finally works!
You can calibrate the bender by letting it in the center and pressing FROM TAPE + 3.

prohet13, 2017-01-25:
I really should read the manuel carfeully before asking stupid questions :D now it works perfect! Thanks a lot !

jxalex, 2017-01-29:
I got the similar problem. With Z80 the synth works, with teensy not. but I have very fine good soldering. After I put the Z80 back, the synth still works.
The installation manual is no help, becouse it was rather meant for those who know barely know which end is hot on a soldering iron.

Where is the all hardware debuging or troubleshooting information?
And can someone explain now whats the trouble why it does not work with e-bay chinese clone teensies?

So where from to start the check? (yes, the +5V arrives to the board).

It would help if atleast there would be some SERVICE MODE or troubleshooting or TEST mode in your firmware. Is it?!

hotswank, 2017-01-29:
Hi, it does not work with China Teensy clones because they are China Teensy clones!!!

Hope that helps.

gligli, 2017-01-29:
It does not work with teensy clones because the bootloader is different and p600fw depends on an undocumented behavior of the official PJRC bootloader to be able to write flash.

If the soldering is correct and it's an official teensy properly flashed with p600fw, it should at least boot and display something.

The firmware can be compiled in DEBUG mode, which will use USB to log stuff but there was never a need for a SERVICE mode.

jxalex, 2017-01-29:
So,  You are using other peoples brainpower but you still drain huge money as you make secrecy of some source code? But this is ridiculous!! This is NOT opensource.

I would have contributed too after I had got running my P600 as I develop also tweaks for other synthesizers as well. I think I will better use Z80 and dont bother about it if you are with such attitude!

jxalex, 2017-01-29:
why I should ever ever consider contributing some work or coding, debuging, if you charge $150 for some kit?!??

gligli, 2017-01-29:
That's not what I said. So you seem to be able to understand code, here's the code that uses the teensy bootloader: https://github.com/gligli/p600fw/blob/master/firmware/teensy_bootloader_hack.h
As you can see, it's not like I had a choice if I wanted to be able to program the teensy flash.

It's clearly mentionned everywhere that this project is based on a $27 teensy++ chip, I get not a single dollar from PJRC.

Also, I'm totally fine with you not using p600fw, and not contributing anything.