gligli/p600fw #94: Error message after everything working fine

AGP84, 2017-02-12:
Hi, I have been using my P600 for the last year with GliGli upgrade without any problems, then one day (after using it the previous day without problem) when turning it on behaved strangely - at first one of the voices seemed to be out of tune but the tuning routine didnt help, and a few cycles of turning off/on ended with it making odd noises and no musical sounds.  I also tried the tuning routine a few times which seemed to be taking longer to complete but didnt solve the problem.  The orange light on the teensy still glowing as normal.  I noticed when the problem started, on powering up the display was scrolling an error message which I think is Uer 2,0992 (it may be 2,092).  I'm pretty sure it didnt display this message before the problem occurred.
Then, after a couple of days not using it, I turned it back on and the error message is still there but everything seems to be working normally - all of my sounds are there and everything in tune, functions all seem to be working, this is great but very concerned by the error message and if it's going to become non-musical again!  I havent been able to find any reference to the error message and what it might mean, do you have any ideas please?

gligli, 2017-02-12:
Hello!
The message is "Ver 2.0992", it's just the firmware version number, it's not an error message :)
The actual problem could be a dry solder joint or a connector / chip socket not making proper contact I think...

AGP84, 2017-02-13:
Thats great news I'll just keep an eye on it and check chip sockets etc if it happens again thank you very much!

closed 2017-02-13 by AGP84