gligli/p600fw #95: No sound?

prioret, 2017-02-15:
I have no sound.

And, when turn the knobs they go from 00 to 5.5 with no numbers in between with the slightest touch.

I just performed the upgrade, sent over the p600firmware_stable_2_0.syx and got an S. Then loaded the Prophet600OriginalFactoryPatches.syx

Any ideas?

prioret, 2017-02-15:
I may have fixed it. I soldered yellow to F0 instead of where it belongs. I fixed that.

BUT

When im in preset mode and select a patch it comes up, but when I start tweaking the knobs, I dont see the knob value on the display.  when I'm out of preset mode, I do see the knob values.

gligli, 2017-02-16:
Hello,
That seems normal about the knob values in preset mode. You can press "TO TAPE" to make preset mode behave temporarily like manual mode.

prioret, 2017-02-16:
Thanks for your quick response and your software. So far so good!