gligli/p600fw #96: Gligli firmware starts up then no response

slowpilgrim1, 2017-02-20:
Hi there,

I just bought my Prophet 600, which came with a Teensy board but the previous owner had chosen not to install it. I was told the firmware had already been uploaded on to the chip, and if I wanted to use it I just needed to install into the synth.

When I insert the chip and switch the synth on, the LED does not light on the board but the synth itself reads the welcome message and indicates ver 2.00. After this, the message disappears and the synth behaves as it if was switched off - no readout, no sound, and the control panel is unresponsive. When I put the Z80 back in, the synth works fine again.

I tried re-flashing the Teensy but it will not talk with any of my PCs, and the LED will not light up even after I tried to put it in Manual Program mode. I don't know anything about electronics or programming, but assume it's not receiving power for some reason. It's odd that it will at least display the welcome message on the synth though.

I don't know if it's related, but the previous owner mentioned that he thought the synth itself was faulty because the sound becomes mangled sometimes with the Z80 in. It also wiped nearly all of his patches, and replaced them all with the same patch which just makes high resonance sounds.

Since I've been playing with it, I've experienced no issues - I can program new patches fine, and they save correctly, and I can overwrite his 'faulty patches'. This makes me think the issue he had was something to do with trying to install the Teensy.

Sorry if this all sounds vague, but I've had it for one day and am not good at trouble shooting. Have you got some ideas on how I can begin to identify and solve the problem?

Thanks for your time,

Ash

slowpilgrim1, 2017-02-20:
board is Teensy++ and appears to be genuine

gligli, 2017-02-20:
Hello,
Try putting the synth in update mode by pressing FROM TAPE + TO TAPE while powering the synth.
If it displays "U", then you could try uploading this version: http://gliglisynth.blogspot.fr/2015/09/p600fw-210-third-release-candidate.html though MIDI sysex. (This is better explained in version 2.0 user manual)

slowpilgrim1, 2017-02-22:
Hi, thanks for the response. I bought an M audio midisport uno and tried that with no luck. I put the synth in upgrade mode as you said, get the 'U', but the moment I connect the MIDI in cable from the M audio to the MIDI in on the synth and run Sysex, I get an E the moment I send. I've tried increasing the delay between buffers to 300ms and also the Input/Output buffer size to 1024 bytes with no luck, always the E.

Is there anything else I can try? Perhaps I need a better MIDI interface, or to reflash the teensy via usb?

Thanks.

gligli, 2017-02-22:
Hmm, I think it seems like a proper midi interface but make sure it's not sending MIDI clock or any other stuff while you send the sysex. The updater has a really basic MIDI parser which would error when receiving that.

slowpilgrim1, 2017-02-26:
Hi, thanks for the reply! I checked and it's not sending MIDI clock or anything else from what I can see.

I tried updating using snoize on my Mac instead of MIDI-OX on my windows desktop and still got the error. However, despite the manual advising not to, I've tried switching it on and now I at least get:

A) welcome msg
B) preset numbers displaying in preset mode
C) jumbled characters which are scrambled when I enter manual mode
D) Parameter names will display correctly on the screen when you key in different settings (but flicker occasionally)

As soon as I hit a key on the keyboard, the screen goes blank and the unit is practically 'switched off' again.

Weird.

felixp600, 2017-03-22:
Hi!  I had the exact problem that Teensy++ 2.0 switched off as soon as I press the key on the keyboard after the welcome message.  And it seemed like that was a bad connection between one of the pins & a CUP socket.

I found that out since the pins on my Teensy++ 2.0 board (purchased directly from PJRC) are too thick to fit into the original Z80 socket!  I pushed it sooo hard, but at one point I thought I'm going to break something, so I decided to solder the board on an extra CPU socket that I had which has small pins.  Then mounted that board + extra socket onto the P600.

It worked fine in the beginning, but suddenly I found it gets switched off.  So I re-soldered the pins again, and it came back.  And once I had to remove this board + socket to check something else, and re-mounted them.  Then the same thing happened again, and I re-soldered the pins, and it's all fine again!

Although MIDI sync doesn't work on mine (other MIDI function works OK so far)...  which I'm currently trying to figure it out.

Hope this helps a little...