gligli/p600fw #98: Help, Gligli firmware upgrade

syntomik, 2017-08-08:
Please help,  pressing "from tape" + "to tape" the "U" appears on the display but when I send the syx file from midiox nothing happens. The syx file is sending but the P600 always shows "U". I'm using a db15-midi cable from the sound card to midi on the P600 ( I don't have a USB-midi cable). Could this be a setting issue in midiox or some other problem? Not sure where to go from here, any help appreciated. thx

hotswank, 2017-08-08:
Have you verified the sound card/midi cable can succesfully send any sysex data to any of your other synths?

syntomik, 2017-08-08:
Naturally, that's the first thing I tried. I successfully transmitted sysex to Siel DK600 using the same method. So, I'm wondering if P600 is waiting for data on the midi in or not? anything else I can check?

syntomik, 2017-08-09:
update: I went out to purchase a USB-Midi interface, installed it, drivers loaded. Still nothing from the P600. Actually, most of the time it stays on "U" but sometimes I get "E". I configured the timing buffers to 250ms like the manual says but still nothing. Still no problems sending sysex to my Siel. Does anyone know what's going on, why I can't upgrade the firmware? Am I overlooking something or setting?  
I tried both p600firmware and p600fw_stable_2.0, no difference
Any suggestions, anyone?

hotswank, 2017-08-09:
Do you have a genuine Teensy board and not a China clone from eBay? Does your current P600 firmware respond to MIDI CC and NOTE on/off correctly? What is your current firmware version and why do you need/want to flash it? Can you succesfully dump a sysex to MIDIOX: 
"Miscellaneous Settings function “6” will send a SysEx dump of all presets on MIDI out. Use your favourite SysEx utility to store it."  http://www.midiox.com/cgi-bin/yabb/YaBB.pl?board=query;action=display;num=1424865245  HTH

syntomik, 2017-08-09:
Yes, I have a genuine teensy with orange led. I'm just following the manual so I flashed the teensy with the hex file, it booted , got the scrolling message and went thru its calibration. Next step, upgrade the firmware, this is where I'm/teensy is stuck.  P600 eprom is six-0-8, I can't seem to upgrade the teensy firmware? I'm not dumping to PC

syntomik, 2017-08-10:
So, re-installing the Z80 and checking midi in/out data with midiox works flawlessly without issues, midi in/out connections are good from p600 to pc, even re-loaded the factory patches without problems.
So my problem lies within the teensy ++ 2.0 
I will start over with the connections and re-flash the teensy and go from there. 
thank you all for your suggestions!

elauria, 2019-03-18:
I know you successfully sent sysex to your Siel, but I'm gonna say this anyways: is the "pass sysex" option in midiox enabled? (Options -> Pass Sysex). I missed this one and was struggling to understand why it wasn't working.