gligli/p600fw #99: Arpeggiator stopped working

Retrofuturist, 2017-08-09:
I did the upgrade a few months ago and it has been great.  My arpeggiator was working just fine up until a few days ago when it just completely stopped.  I turn it on and I get no sound from any keys, dead silence from all voices.  Doesnt matter which arp pattern i use, dead silence when the button is active.  Switch it off and the voices are back.  Not sure which version of firmware in using, I'll check later today.

braincloudsound, 2017-12-13:
I'm having the same issue.  Did you find a solution?

amichaela23, 2017-12-18:
Anything come of this? I just got this thing got one and update the OS to the 2.1 to get the sequencer to work and now neither is working!

braincloudsound, 2017-12-18:
Not yet. I just ordered a z80 processor chip on the advice of a tech. Since my prophet 600 still has the original rom/ram chips installed, apparently I can remove the gligli update and replace the z80 and it should function as it did originally. If at that point the arpeggiator still doesn’t work, then the problem isn’t with the gligli. It’s a separate issue that will need to be addressed. If it DOES work however, then it’s likely my gligli chip is faulty and I will try reflashing the OS and trying it again.

amichaela23, 2017-12-18:
Ugh, I just bought this thing and the arpeggiator went up when I did this OS update, trying to get the sequencer to work. My tech is gonna be mad at me probaby for buying it with the gligli update haha.

amichaela23, 2017-12-18:
How do yo u"reflash" the OS?

braincloudsound, 2017-12-18:
Yeah, mine was a new purchase. Should have demoed the arpeggiator before I left.

Anyway, I don’t know how to reflash the os myself. But basically, there is a USB port on the gligli chip which is how the os is loaded and I would have my tech reload the os onto it again just to make sure it’s fully functional or replace the chip completely with another gligli chip. There is a guy who goes by synthchaser in San Diego who is helping troubleshoot this with me.

amichaela23, 2017-12-18:
Might be worth a drive to Sand Diego cause I'm really scared my keyboard tech is gonna get mad at me cause I bought it without talking to him first, haha.

braincloudsound, 2017-12-18:
Yeah, I’m in LA. Weird we are nothing having similar problems. Other than the seq and arp issue, I love the upgrade. The faster envelopes and dual vco levels that can drive the filter are enough to make it worth it Aline. But I was really looking forward to the arp. 

I’m waiting on the z80 from China. Mistakingly bought one before checking the vendor so it will likely be a week or so before I can deal with it. 

The guy in San Diego told me he comes up to pick up drop off every so often so it might all work out if you’re not in a huge hurry

Scot
323-823-2031

Retrofuturist, 2017-12-18:
My arp still isn't working.  I don't want to revert to the Z80 because I have several great patches that I programmed with the new chip that I don't want to lose.  Did either of you have success refreshing the OS?

amichaela23, 2017-12-18:
No luck, I’m taking it in to a tech today. Did you update to the 2.1.1? Did your arp ever work with 2.0?

There’s always tape!! :)

braincloudsound, 2017-12-18:
no, I literally have only owned it for 6 days.  I got it with the Gligli installed and it says 2.0 on boot up.
I hope the Z80 chip gets here this week so I can swap it out and see if it works in the original state.
If not, I’m unsure whether that is something easy to fix, or if it would be too expensive to troubleshoot.  

But let me know what your tech says.  Could be helpful to narrow down the trouble spot on my p600.

Scot Lang
(323) 823-2031
www.braincloudsound.com <http://www.braincloudsound.com/>

amichaela23, 2017-12-19:
Okay, I dropped in a z80 and the arpeggiator works. My panel needs to be replaced for the sequencer to work. The tech I went to is not a fan of the gligli but I also inhabit a more traditional music world. I’m keeping the z80 getting the panel replaced and keeping it OG. I wanted a John carpenter machine basically.

amichaela23, 2017-12-19:
Also got to borrow the original owners manual which is pretty Stoney

phawsea, 2017-12-29:
I think I do have the same issue. The arp should work as soon as I hit one of the two arp buttons shouldn't it? As I can see when starting up I should have version 2.099 installed (I didn't do it myself).

braincloudsound, 2017-12-29:
Yes, it should. I’m still waiting on a z80 chip so I can restore the p600 to its standard operation to see if the arp and seq start working again. Seems you are the 4th person with a similar problem. Seems that some gligli updates are problematic. I bought my p600 with the upgrade preinstalled so I’m not certain who supplied the update but it’s possible these are all from a bad batch. 
Fortunately there is a guy near me who sells the upgrade and if the z80 swap does the trick, I might be able to get the upgrade from him with a money back guarantee if his os doesn’t work. Will update soon.

Retrofuturist, 2017-12-29:
I built my chip from scratch and flashed it myself, so maybe the firmware version I'm using is unstable.

amichaela23, 2017-12-29:
I still need mine serviced to get the seq to work. I’m really digging the sounds with the Z80 tho. I don’t really have a gligli relationship cause I took it right in. 

Are the sounds different?

jhhl, 2017-12-29:
GliGli's mod greatly expands the capabilities fo the old P600. The oscillators and filters are still analog, but the digitally created envelope generators, LFOs, and MIDI controls have been much improved! The Arpeggiator is also different from the original, and there's a step sequencer that I don't like very much.
 
I'd give more advice, but my P600 us unwell, unrelated to GliGli issues.

braincloudsound, 2018-01-13:
A little update for those interested. I reverted to the Z80 chip and everything started working again.  I asked a tech about a possible problem with the Gligli update itself, and he explained there was a version where the arp and seq functionality was removed because he needed the memory for other functions, but was supposed to put it back (and it was) in a later version.  I can't confirm which version since I removed the update, but I am looking to have it flashed and updated to the latest (working) version with arp and seq functionality.  If anyone can confirm which version this is.  Thanks