gligli/p600fw #103: Firmware beyond 2.0 stable

ceharris, 2018-04-04:
Looks like some of the MIDI features I want (e.g. PATCH DUMP REQUEST) are not in the stable 2.0 release. Is there a later version I can run? Do I have to build it myself? Will there ever be another "stable" version generally available?

hotswank, 2018-04-05:
I really would like to know this as well, but have you tried any of the later builds and ran into problems or what?

hotswank, 2018-12-09:
Any news? I'm depending on the patch dump request because I build my own editor in Max and without that patch request I've wasted many many hours building something I can't finish.

gligli, 2018-12-13:
Yes there is, you can look at my blog for the lastest version: https://gliglisynth.blogspot.com/search/label/p600fw
(It would be P600fw 2.10 third release candidate. 2.10 was never released because I almost lost interest in the project)