gligli/p600fw #107: 2 Octave shift for MIDI in

sounddoctorin, 2018-09-05:
I have no idea what caused this.  I'd been using the upgrade successfully including with MIDI in with my sequencing computer.  All else responds ok but P600 is 2 octaves higher now.   I see no way to midi map in the upgrade so I'm lost.

Jpro600k, 2019-03-17:
You know press twice 'from tape' button.
press key(ex: c or d or e or ......)
This means that you can change the octave.
The lowest key will be -24.
The highest key will be +36.  
amazing.

sounddoctorin, 2019-03-17:
Jpro thanks for getting back.   First they seem to have left out any hint of how to know version number in manual also.   Stuff scrolls by so fast i can't read it on boot.   Maybe that's a hint?  did that get slowed down so humans that didn't write the code can read it?  hehe.  Anyway hitting 'from tape' twice quickly seems to do nothing different... i hit a note...no change in 2 oct. midi offset

sounddoctorin, 2019-03-18:
Oh I see it says beta 1 i think