gligli/p600fw #117: Do i have a dud?!

TheRaydevil, 2023-02-04:
Hey everyone, My nice GliGli upgrade is unfortunately not going as planned... Bought a Teensy++2.0 from Tinkesphere, Prepped it as per instructions, loaded the 2.0V hex file, installed it... power on and... Nothing. Took it out again and made sure to check everything top to bottom, over and over. Everything in order. re-loaded the file a couple of time... still nothing. Finally Decided to try loading it with Imogens version 2.1, Re-installed it, power on and now i have the tuning sequence starting but crashes right after going thru the filters leaving the number 6 on the display. It just freezes like that. Tried over and over, even the GliGlis 2.0V one last time. Nothing.

I can only get it to "power up" with Imogens version but then of course it freezes. Any other version will do nothing. If i put the original Z80 back in everything works normal. I sort of lost now... Any ideas?

matrix12x, 2023-02-04:
Did you correctly do the mods to the teensy? carefully go over to make sure the solder joints look good.

Also,  if you got a teensy clone, it probably will not work. The bootloader on the clone is wrong and the Gligli software relies on the bootloader.

TheRaydevil, 2023-02-05:
Thanks for the reply! I've gone through everything over and over. Everything looks fine. Joints are nice, everything is routed correctly. I got the Teensy board from Tinkersphere (actually followed the link posted directly on the PJRC website) so i'm hoping i didnt get a clone. LED is Orange... as i read most clones are various other colours. 
Strange thing is that if i load any other versions other than the Imogens 2.1 the LED stops working. If i plug it back to the computer i get nothing. no blinking... Same when i put it in the Synth. Dead... nothing.

image-et-son, 2023-02-06:
[image]

Do you mean this link? Even though it's been linked from PJRC, I am not convinced that these are actually original PJRC boards. Why can you preorder them? They've out of production for a long time. Sounds very strange. 

And I agrre, it is most likely that you are working on a board with a bootloader that is different from the one PJRC is using. In that case, the firmware does not work.

I am still working on making the firmware work on other boards, but the outcome is totally open.

TheRaydevil, 2023-02-06:
Yup!! That would be the one. That "Pre-order" thing is new i guess. When i bought mine they had a couple in stock. I was stoked to have found some and was pretty excited to receive it and start the project but now i see THIS... Which totally makes me think that the one i received is a stupid knock off.... 100$ Canadian including shipping for this...

TheRaydevil, 2023-03-19:
I wrote to Tinkershere asking about that actually. Their answer was to "test the board" by trying the Slow blink Fast blink hex files to make sure my unit wasn't defective. (which i did and everything turned out fine) But they did answer the "being a clone" question.

TheRaydevil, 2023-03-19:
And i see now they they (Tinkersphere) has them back in stock now... So we most certainly know now that these are indeed clones

Midphase, 2025-08-29:
Genuine Teensy 2.0 ++ have become unobtanium in 2025. Any possibility of making this work with counterfeit clones since that's all that is available? I too am getting stuck after the tuning routine where the display only reads 6 (or is it supposed to be G from the welcome scroll?).

Anyways, any help is much appreciated here, at the moment I fear that without a solution to run this mod on a counterfeit clone, or alternatively to port it to a newer Teensy, the GliGli Prophet 600 mod is effectively dead.

matrix12x, 2025-08-29:
Are you running this version: 
https://www.image-et-son.com/

Also, some P600s have a mux chip that over time is slow and sometimes leads to these tuning issues. 
https://prophet600revisited.blogspot.com/2022/06/feedback-on-2022-beta-watch-your.html

Midphase, 2025-08-29:
Yes, I tried it both with the latest GliGli Stable 2.0 and the latest Imogen 2022_1, on the GliGli nothing happens, like no display no nothing. On the Imogen version it goes through the Tuning Routine, after it finishes that, a "6" appears on the display and the unit is effectively frozen. If I hit the Tape In/Out buttons I can get it to go into Update mode, but beyond that nothing happens.  

I did read the blog about the 4051's and this unit has the original factory TI 4051BE multiplexers...

Any other ideas? Does the EPROM Firmware revision matter?

Midphase, 2025-09-17:
What about the idea of completely porting to a non-Teensy solution like it has been posted on GearSpace? Would that turn out to be a less demanding solution with ultimately more longevity?  https://gearspace.com/board/electronic-music-instruments-and-electronic-music-production/845271-prophet-600-users-have-upgrade-144.html#post17565851

image-et-son, 2025-09-17:
@gligli : have you ever used USB storage with the Teensy? There are functions for this but I could never get it to work (probably my fault). It just sounds like an easy way. Having said that, I didn't even know there's an eeprom...

Colpocorto, 2025-09-18:
> [@gligli](https://github.com/gligli) : have you ever used USB storage with the Teensy? There are functions for this but I could never get it to work (probably my fault). It just sounds like an easy way. Having said that, I didn't even know there's an eeprom...

Better let's drop the Teensy forever.

We neee a PJRC-free solution.