gligli/p600fw #118: booting stops with "F"

mpuetsch, 2024-12-18:
Hallo ,
-when I flash the firmware versions
p600firmware_beta_1_0 
and 
P600_firmware_v2022_1
on the teensy xx2, the Prophet start runs as described: 
The orange LED on the Teensy will light and the tune routine will commence stepping
through the 12 oscillators and 6 filters; during the tune routine the display cycle A1 through A6, b1 through b6 and F1 through F6. 
**but** instead of: "After approximately 20-30 seconds, a welcome message will scroll."
**it comes: "F"**

-when I flash the firmware versions
p600firmware V2_10 RC3 from_gligli_Rep_2015-09-05
and
p600firmware_stable_2_0
the prophets doesn´t boot at all.

has someone an idea for this?
what I´m doing wrong?
thank you soo much
Geo from Cologne Germany

hotswank, 2024-12-20:
I'm not an expert but it's common sense to check your power supply and replace caps of your P600. Those puppies are getting very old you know and chances are it's something hardware related. Good luck and I hope you will sort it out!

gligli, 2025-09-18:
This is basically the same as issue #117 
Teensy++ 2.0 that are available nowadays don't have the same bootloader as the original PJRC's and that's why it crashes just after the tuning procedure.
In any case, I'm trying to devise a fix.

mpuetsch, 2025-09-18:
Thank you so muchVon meinem iPhone gesendet